slimcdk/esphome-custom-components #2: TMC2209 esphome component

kdrost1, 2024-09-02:
Hi,
Thank you for theTMC2209 esphome component, done some testing durring the weekend and I don't think the setting :

`rms_current_hold_scale: 0% `

Is working, I noted the TODO text.

/kenneth

slimcdk, 2024-09-03:
Thanks for taking your time to test! I'll take a look at it in the next couple of days. Some other parameters for standstill and current also has to be exposed.

You can interact directly with the registers and fields using [these methods](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#advanced). Don't hesitate to share your findings here, if you decide to experiment/debug further. I appreciate it!

slimcdk, 2024-09-09:
I've decided to remove the option for now as I still need to figure out the best way to expose the run and hold parameters when adding breaking (and freewheeling) functionality. If you have any suggestions or ideas, please let me know.

lexidevs, 2024-09-11:
writing TMC2209_FREEWHEEL_FIELD=3 (%11) and  TMC2209_IHOLD_FIELD=0 doesn't seem to work to enable braking -- in fact, when I use those, the motor doesn't function at all. I'll be trying more things soon and do my best to update here

slimcdk, 2024-09-11:
I'm not 100% sure but it likely collides with disabling the driver on ENN. Could you try connecting ENN from the TMC directly to VSS/GND such that it is constantly enabled?

lexidevs, 2024-09-11:
Yep, I'll try that when I get a chance!

lexidevs, 2024-09-23:
Sorry for the long wait time, I was working on transferring my previous prototype board (NodeMCU which had a lot of issues) to a esp32 board. Here's what I've found: 

Directly setting those registers while hardware enabled does work, there is noticeably more holding force than with TMC2209_FREEWHEEL_FIELD set to 0, but it may still not be enough for some applications. I found that in my application (roller blinds), the force of gravity did sometimes overcome the passive braking force, so I have switched to using the holding current by setting TMC2209_IHOLD_FIELD to some number greater than 0 (6 seems to work fairly well for me).

lexidevs, 2024-09-23:
I think you could expose these parameters in configure with parameters such as `freewheel_mode: int (0-3)` and `rms_current_hold_scale` (which is just a percentage of the rms_current setting and therefore would require that to be set?). You could possibly make it so that when freewheel_mode is set, IHOLD is always set to 0 and overrides the hold current parameter, I'm not sure.

lexidevs, 2024-09-23:
Also: may be helpful to add a note about disabling the logging output to the UART in use for the stepper on certain ESP32 boards. On an ESP32-S3, setting the eFuse `EFUSE_UART_PRINT_CONTROL` to 2 (%10) fixed some problems, but on most ESP32 series chips there is some way to disable the ROM printing over UART during boot (either via an eFuse or a strapping pin) and this may resolve issues. It's possible that the TMC2209 was trying to interpret the ROM log output as commands and, as they were not valid (of course), got into a weird state where it did not respond to commands.

slimcdk, 2024-09-25:
Hi @lexidevs,

I've made some changes and added a few features like holding current and standstill mode. Readme isn't complete for the changes yet, but please take a look at the branch https://github.com/slimcdk/esphome-custom-components/tree/better-tmc2209/esphome/components/tmc2209

Good point on the UART. Up untill now I only used free pins for the serial connection.

slimcdk, 2024-09-27:
IHOLD / hold current is now fixed along some major changes to the implementation. Please check https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#tmc2209configure-action

kdrost1, 2024-10-02:
IHOLD works for me, my application is also roller blinds - Thank you!

closed 2024-10-02 by slimcdk