slimcdk/esphome-custom-components #4: TMC2208 support

morotsgurka, 2024-09-04:
Hi again!

I've tested the TCM2209 component with a TMC2208 driver. This seems to work fine, although Im getting a lot of stalls whatever i do?
The IC detected is 0x20: [tmc2209:025]:   Detected IC version: 0x20

So potentially adding "support" for 2208 could be as simple as creating another README with information about the TMC2208:
- You have to bridge the solder pads on the underside to use UART mode
- You have to wire TX and RX into a single wire to the driver with a 1K resistor like [explained here in step 6](https://www.instructables.com/UART-This-Serial-Control-of-Stepper-Motors-With-th/)

slimcdk, 2024-09-04:
Awesome to hear! I don't have any tmc2208s myself, but I'll look into the possibilities of supporting tmc2208. 
tmc2208 is lacking some core features like stall detection, so it would probably be it's own component / implementation, but I'm open for a tmc220x component with type selection.

closed 2024-09-13 by slimcdk

slimcdk, 2025-02-12:
Hi, I've created a component for tmc2208 https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2208/README.md