slimcdk/esphome-custom-components #6: Stealthchop maybe?

swifty99, 2024-11-14:
This looks brilliant. I will test it as soon as I can.

the esphome AC dimmer has hardware interrupts enabled. If you venture in more accurate timing. I am also working on a driver with precise timing. Progress is slow, unfortunately.

This is no issue, however I chose the Trinamics for their ability to do stealth chops. My machines have to be quiet. Sensorless homing is not possible at the same time. I changed modes for that on a bare metal project. I would love to see this extended.

Great to see progress here!

slimcdk, 2024-11-14:
Thanks! 

Are you referring to using interrupts for feedback from the TMC2209s internal step generation for position tracking?

If you are looking for looking for particular configuration options, please take a look at this ["advanced mode"](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#advanced)

swifty99, 2024-11-18:
Hi,

Here it is well explained: https://youtu.be/Q0sJlGh9WNY?si=cUkBXtGcYgAoYZl-

stepper motors can be super silent!

slimcdk, 2024-11-19:
Running the motor in stealthchop mode is done through setting threshold values for a set of registers. Are there any particular registers you need? I'll provide more concrete examples if you tell me which. For stealthchop tuning you must create a configuration that goes through the steps mentioned in the datasheet.

I'll work towards exposing more registers as native esphome actions at some point, but all registers are read and write accessible through the ["advanced mode"](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#advanced) for now. Use them in a lambda context and you should be good to go.

> the esphome AC dimmer has hardware interrupts enabled. If you venture in more accurate timing.

Also, could you elaborate a little more on this? The TMC2209 component uses interrupts for position tracking when in [uart control mode](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#control-the-position-using-serial-uart). The pulse train generated by ESPHome could however benefit from using hardware timers to generate steps when using [that mode](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#control-the-position-using-traditional-stepping-pulses-and-direction).

swifty99, 2024-11-20:
Hi,

It has been a while since I used stealthchop. I was using AccelStepper library, and changing modes with my own routines. Accelstepper has functions like: `driver[driverNo].stealthChop(1); `  
Not a big priority right now, however would be nice to have an ESPHome "setMode" command or alike. Advanced register changes would probably possible to do the same.

The AC dimmer component needs super strict timing to set outputs according to AC phases. They use a HW timer after the AC phase changed to set the powerswitch.
There is this PR, which updates this to IDF: https://github.com/esphome/esphome/pull/7072

The timer handling is a bit complicated (for me) as the always global interrupts need to call protected ESPHome component objects.

swifty99, 2024-11-20:
Hi,

you are right, it was TMCStepper.
And yes hardware timer are super delicate, break with every IDF update and so on.
I have not tried the UART mode yet (don't change a running system). But I will definitely do! It looks very promising. I do not have super accurate stepping requirements most of the time.

Thanks for your driver. There are some stacked projects which will definitely benefit from this.