slimcdk/esphome-custom-components #8: Direction + posistion

kdrost1, 2024-11-27:
Hi,

Just compiled TMC2209 with esphome 2024.11.1, was previously on 2024.10.3.
I'm using TMC2209 to control my blinds
Now the direction is wrong and feedback from stepper is broken.
tried using "drirection" under tmc2209.configure.

`- tmc2209.configure:
            microsteps: 8
            direction: counterclockwise`
            
This makes the stepper motor run forever:

` - platform: template
    name: 1000 Steps forward
    on_press:
      - stepper.set_target:
          id: driver
          target: !lambda return id(driver).current_position +1000;
`

When is runs, motor is a lot slower than normal,

slimcdk, 2024-11-27:
Hi! There were some breaking changes. Top part of the [README](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md) has a solution if you want to continue with the previous version.

A few things regarding the speed/direction calculation were wrong, so they are now corrected. Direction has been reversed and speeds are about 71% slower . You can achieve smae speeds if you increase yours with 40% (`speed * (0 / 0.715)`)

Can you share your config? I'll take a look to see if there is an issue.

kdrost1, 2024-11-27:
Here you go, i'm no programmer so please be gentle :-)

```
esphome:
  name: roller-south
  friendly_name: Rullegardin
  platformio_options:
    board_build.flash_mode: dio
  on_boot:
    - priority: 800.0
      then:
        - tmc2209.configure:
            microsteps: 8
            direction: counterclockwise
        - tmc2209.currents:
            #stallguard_threshold: 25
            standstill_mode: normal
            irun: 16
            ihold: 4
            #run_current: 800mA
            #hold_current: 0mA
            tpowerdown: 0
            iholddelay: 0
        - tmc2209.stallguard:
            threshold: 25
    - priority: -100.0
      then:
      - tmc2209.currents:
          ihold: 4
      - stepper.report_position: # Set stepper to global variable
          id: driver
          position: !lambda return id(my_stepper_global);
      - stepper.set_target: # Set stepper to global variable
          id: driver
          target: !lambda return id(my_stepper_global);
      - if: # If blind is Closed
          condition:
            - lambda: 'return id(my_stepper_global) == 0;'
          then: # Publish state etc.
            - cover.template.publish:
                id: blinded
                state: CLOSED
                current_operation: IDLE
      - if: # If blind is Open
          condition:
            - lambda: 'return id(my_stepper_global) == id(endstop);'
          then: # Publish state etc.
            - cover.template.publish:
                id: blinded
                state: OPEN
                current_operation: IDLE
      - if: # If blind is Neither
          condition:
            - lambda: return (id(my_stepper_global) != 0) && (id(driver).current_position != id(endstop));
              #'return (id(my_stepper_global) != 0) && (id(driver) != id(endstop));'
          then: #  # Publish state etc.
            - cover.template.publish:
                id: blinded
                position: !lambda 'return (float(float(id(driver).current_position) / float(id(endstop))));' 
                current_operation: IDLE

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
[... 170 lines cut ...]
    name: 1000 Steps forward
    on_press:
      - stepper.set_target:
          id: driver
          target: !lambda return id(driver).current_position +1000;

  - platform: template
    name: 1000 Steps backward
    on_press:
      - stepper.set_target:
          id: driver
          target: !lambda return id(driver).current_position -1000;

  
  - platform: template
    name: IHOLD_IRUN
    update_interval: 10s
    min_value: 0
    max_value: 31
    step: 1
    lambda: return id(driver)->read_field(IHOLD_FIELD);
    set_action:
      - lambda: id(driver)->write_field(IHOLD_FIELD, x);

```

slimcdk, 2024-12-05:
Can you remove priority for the `on_boot` actions? The priority 800 you are using is nearly before the TMC component is ready.

Something like this should be good enough:
```yaml
  on_boot:
    - tmc2209.configure:
        microsteps: 8
        direction: counterclockwise
    - tmc2209.currents:
        #stallguard_threshold: 25
        standstill_mode: normal
        irun: 16
        ihold: 4
        #run_current: 800mA
        #hold_current: 0mA
        tpowerdown: 0
        iholddelay: 0
    - tmc2209.stallguard:
        threshold: 25
```

Moreover, the `IHOLD_IRUN` config at the very bottom is a sensor config and doesn't belong under button. 

Let me know if you continue to have issues or I'll close the issue after the weekend.