slimcdk/esphome-custom-components #9: Rotate motor

Littleelee, 2024-12-05:
Tell me please, how to make the motor just rotate infinitely at a given speed?
I use control via index_pin.

slimcdk, 2024-12-05:
Hi,

The TMC2209 offers functionality to essentially run forever at a certain speed, but that doesn't fit well in the context of the ESPHome stepper implementation with position tracking. I have however thought of implementing a way around it, but haven't looked much into it yet.

You can tell the driver to move forever by [writing to](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#advanced) `VACTUAL_FIELD`. It won't track the position however.

Littleelee, 2024-12-12:
Thanks. It works.

slimcdk, 2024-12-12:
Great to hear - I've got a draft for velocity control, but not anything that is ready to be used yet.

Do you mind to share your setup/config here - for me and others to see how you utilise the component?