slimcdk/esphome-custom-components #10: Dual uart support for tmc2209

Prozpect, 2025-01-09:
I have a board where the dual build in tmc2209 drivers share the same uart.
Are you planning on adding support? 
That would be much appreciated! 

I have it working using Arduino, but ESPHome is a much cleaner solution. 

Awesome work you have done!

slimcdk, 2025-01-10:
ESPHome enforces the philosophy about UART being a point-to-point interface. Meaning, multiple components can't use the same UART. Control of multiple drivers would have to be worked around directly by the component itself which would break other inherited concepts from the general stepper implementation. It's sadly not feasible at this point or the near future. Two TMC2209 drivers require two separate UART connections.

What board do you have? Does it use a bidirectional communication with adressing or multiplexing? And if you don't mind, could you share your Arduino sketch?

slimcdk, 2025-01-24:
Hi @Prozpect,

Please take a look at the [`tmc2209_hub`](https://github.com/slimcdk/esphome-custom-components/blob/multiple-tmc2209/esphome/components/tmc2209/README.md#configuration-of-hub). It's on another branch for now as testing, so make sure it is the correct one when reading the docs. I've only briefly validated controls using serial, which has worked fine. Let me know if you encounter any issues or clarification on setup.

Prozpect, 2025-01-24:
Awesome! I hope i have some time tomorrow to test this, keep you posted.

Prozpect, 2025-01-25:
Hi @slimcdk , i tried tmc2209_hub, but I get an error about no CONFIG_SCHEMA.
Any suggestions? Do I need to update anything?

The message

> INFO ESPHome 2024.12.4
> INFO Reading configuration /config/esphome/esphome-web-e7b4f4.yaml...
> Failed config
> 
> tmc2209_hub: [source /config/esphome/esphome-web-e7b4f4.yaml:41]
>   
>   Component tmc2209_hub cannot be loaded via YAML (no CONFIG_SCHEMA).
>   id: hub1
>   uart_id: comms

The code

```
esphome:
  name: esphome-web-e7b4f4
  friendly_name: ESPHome Web e7b4f4
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: 
  password: 
  manual_ip:
    static_ip: 
    gateway: 
    subnet: 

external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209_hub, tmc2209, stepper ]

uart:
  id: comms
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 500000

tmc2209_hub:
  id: hub1
  uart_id: comms

stepper:
  - platform: tmc2209
    id: driver1
    tmc2209_hub: hub1
    address: 0X00
    step_pin: GPIO14
    dir_pin: GPIO12
    diag_pin: GPIO36
    max_speed: 2500 steps/s
  - platform: tmc2209
    id: driver2
    tmc2209_hub: hub1
    address: 0X01
    step_pin: GPIO25
    dir_pin: GPIO26
    diag_pin: GPIO39
    max_speed: 2500 steps/s

```

Prozpect, 2025-01-25:
And on the github in this example the hub points to uart1, but shouldn't this be tmc_comms?

```
uart:
  - id: tmc_comms
    tx_pin: ...
    rx_pin: ...
    baud_rate: ...

  - id: uart_for_other_stuff
    tx_pin: ...
    rx_pin: ...
    baud_rate: ...

tmc2209_hub:
  id: hub1
  uart_id: uart1

stepper:
  - platform: tmc2209
    id: driver1
    tmc2209_hub_id: hub1
    address: 0x00
    ... all other options

  - platform: tmc2209
    id: driver2
    tmc2209_hub_id: hub1
    address: 0x01
    ... all other options
```

slimcdk, 2025-01-25:
Correct, it should be `tmc_comms` and not `uart1`. The external component should point to the `multiple-drivers` branch.

```yaml
external_components:
  - source: github://slimcdk/esphome-custom-components@multiple-drivers
    components: [ tmc2209_hub, tmc2209, stepper ]
```

I'll update the docs!

Prozpect, 2025-01-25:
i tried the other source

```
external_components:
  - source: github://slimcdk/esphome-custom-components@multiple-drivers
    components: [ tmc2209_hub, tmc2209, stepper ]
```
 but still the same error.

I tried clean building files, and the same error shows.

> INFO ESPHome 2024.12.4
> INFO Reading configuration /config/esphome/esphome-web-e7b4f4.yaml...
> Failed config
> 
> tmc2209_hub: [source /config/esphome/esphome-web-e7b4f4.yaml:41]
>   
>   Component tmc2209_hub cannot be loaded via YAML (no CONFIG_SCHEMA).
>   id: hub1
>   uart_id: comms

I also did a clean install of esphome builder just to be sure, but same result.

slimcdk, 2025-01-25:
Can try to set [`refresh`](https://esphome.io/components/external_components.html#refresh) to `0s` for the external component? It might use a cached version and this will force it to fetch the latest.

Also, can you share your config?

Prozpect, 2025-01-25:
have to do some other things now, I'll get back to you later today

Prozpect, 2025-01-25:
Tried adding refresh and now the error indicates:  "couldn't find remote ref multiple-drivers"

code
```
external_components:
  - source: github://slimcdk/esphome-custom-components@multiple-drivers
    components: [ tmc2209_hub, tmc2209, stepper ]
    refresh: 0s
```

Error message:
> INFO ESPHome 2024.12.4
> INFO Reading configuration /config/esphome/esphome-web-e7b4f4.yaml...
> INFO Updating https://github.com/slimcdk/esphome-custom-components.git@multiple-drivers
> Failed config
> 
> external_components: [source /config/esphome/esphome-web-e7b4f4.yaml:31]
>   - 
>    couldn't find remote ref multiple-drivers.
>     source: github://slimcdk/esphome-custom-components@multiple-drivers
>     components: 
>       - tmc2209_hub
>       - tmc2209
>       - stepper
>     refresh: 0s

slimcdk, 2025-01-25:
I see what's wrong.. I wrote you the wrong branch name! The branch is named `multiple-tmc2209

Prozpect, 2025-01-25:
Awesome, it's working now!!
I see both steppers are communicating through the UART.
Now the next step is to get all the functions into the code like stallguard, but I'm not worried about that, the biggest hurdle has been taken.

Thank you so much for your effort. I'm going to buy you a cup of coffee.
Enjoy the rest of the weekend. 
Cheers

slimcdk, 2025-01-26:
Thank you very much! I'll merge the `multiple-tmc2209` branch into master sometime next week.

closed 2025-01-26 by slimcdk

slimcdk, 2025-01-27:
Fix is now on master branch.

Prozpect, 2025-01-27:
FYI

I initially had problems with the steppers stopping around every second with a fixed interval and making a ticking sound in the same rhythm, even when not moving.
Turned out that removing the refresh 0s was the solution to solve this.