slimcdk/esphome-custom-components #11: StallGuard

1iverea9er, 2025-01-10:
Hi
I can't run the StallGuard function.

My driver outputs `SG_RESULT` values from 0 to 90 (approximately). And the value 0 appears anyway at some point. And accordingly, the load is calculated at 100% (the value corresponds to your [readme.md](https://github.com/slimcdk/esphome-custom-components/tree/master/esphome/components/tmc2209#readme)). This is causes the motor to stop.

I also noticed that according to the dataset, the condition for triggering the StallGuard function is the following formula:
 `SG_RESULT = SGTHRS*2`. 
You have a different one:
`(510 - SG_RESULT) /( 510 - SGTHRS * 2) = load coefficient`.

I have a few questions.
Could my problems be related to a specific modification of the driver that I used? 
Why do I get `SG_RESULT` values no more than 90 (approximately). According to the dataset, it varies in the range 0...510?
Why does the `SG_RESULT` value drop to 0 in my config?

My config:
```
external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209, stepper ]

esphome:
  name: trv-exp-stepper
  friendly_name: TRV EXP Stepper
  on_boot:
    - tmc2209.configure:
        microsteps: 8
        interpolation: true
        enable_spreadcycle: false
    - tmc2209.stallguard:
        threshold: 50
    - tmc2209.currents:
        standstill_mode: freewheeling
        irun: 16
        ihold: 0
        tpowerdown: 0
        iholddelay: 0

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "1YlLsOtaOxaYvBbvAQwkclnZrVS1Ogy4hjadXzG2OX8="

ota:
  - platform: esphome
    password: "e5a1ea22b432c8c9339392805bd7f541"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.2.70
    gateway: 192.168.2.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Trv-Exp-Stepper Fallback Hotspot"
    password: "jYi8UhvVh7ek"

captive_portal:

uart:
  tx_pin: GPIO22
  rx_pin: GPIO23
  baud_rate: 500000 # 9600 -> 500k

button:
  - platform: template
    name: Stop
    on_press:
      - stepper.stop: driver

  - platform: template
    name: 1000 Steps forward
    on_press:
      - stepper.set_target:
          id: driver
          target: !lambda return id(driver)->current_position +1000;

  - platform: template
    name: 1000 Steps backward
    on_press:
      - stepper.set_target:
          id: driver
          target: !lambda return id(driver)->current_position -1000;

number:
  - platform: template
    name: Target position
    min_value: -100000
    max_value: 100000
    step: 100
    lambda: return id(driver)->current_position;
    update_interval: 1s
    set_action:
      - stepper.set_target:
          id: driver
          target: !lambda "return x;"

  - platform: template
    name: StallGuard Threshold
    min_value: 0
    max_value: 255
    optimistic: true
    step: 1
    set_action:
      - tmc2209.stallguard:
          threshold: !lambda "return x;"

stepper:
  - platform: tmc2209
    id: driver
    max_speed: 1500 steps/s
    acceleration: 250 steps/s^2
    deceleration: 250 steps/s^2
    #address: 0x00
    enn_pin: GPIO19
    diag_pin: GPIO33
    #index_pin: GPIO32
    step_pin: GPIO5
    dir_pin: GPIO18
    rsense: 110 mOhm
    vsense: False
    ottrim: 0
    on_stall:
        - logger.log: "Motor stalled!"
        - stepper.stop: driver

sensor:
  - platform: tmc2209
    type: stallguard_result
    name: Driver stallguard
    update_interval: 250ms

  - platform: tmc2209
    type: motor_load
    name: Motor load
    update_interval: 250ms

  - platform: tmc2209
    type: actual_current
    name: Actual current
    update_interval: 250ms

```

I used this driver
[image]
[image]

slimcdk, 2025-01-11:
Hi! The selected driver should be fine! Make sure there aren't any bad/loose connections to the motor as that will affect the current/load readings. What motor are you using?

The stall detection formula you see in the codebase calculates a 0% to 100% load from `SG_RESULT` and `SGTHRS`. It is similar to what you have from the datasheet except that they don't calculate the load coefficient but simply outline how to use the values to measure a stall. I will go through some of it at some point as I think there could be edge cases, for instance when `SGTHRS` is zero. In the example config you are setting the threshold (`SGTHRS`) to 0.

I've noticed that it is really difficult to get reliable load/stall measurements using pulse train (step/dir) mode due to how the steps are generated. They are far from consistent and in my experience/setup causes false stall triggers. I'll have to implement hardware timers for pulse generation at some point. Can you try with [control over serial](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#control-the-position-using-serial-uart) to see if that makes a difference?

1iverea9er, 2025-01-19:
Hi!
I'm sorry that it took me a long time to provide feedback. I was busy looking for the causes of the problem and checking your recommendations. So my conclusions are as follows:

1) The driver I chose turned out to be really unsuccessful. I have not been able to adequately control the motor with it. The driver from BigTreeTech (v1.3) was the best choice for me. The MKS driver (v.2.0) also performed well.

2) I also had to do some additional initial driver configuration. In particular, after turning on the sensors, I saw that the values of `TCOOLTHRS` and `TPWMTHRS` have inadequate negative values. I have included the following code in the script that runs with the ESP boot (This setting may need to be included in your code as the default settings):
```
    - lambda: id(driver)->write_register(TPWMTHRS, 0);
    - lambda: id(driver)->write_register(TCOOLTHRS, 1048575);
```

3) I took your advice and switched to UART control.

After these manipulations, the StallGuard function worked normally. 

Thank you so much for your work.

slimcdk, 2025-01-24:
Hi @1iverea9er,

Glad you resolved the driver issue!

`TPWMTHRS` and `TCOOLTHRS` can be configured through [`tmc2209.configure`](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#tmc2209configure-action) as well. I think those values should be configured by the user rather than the implementation. However I don't think those values should appear negative. I'll look into it to make sure they aren't getting casted/read incorrectly.

Thank you for taking your time to look into the issues!

closed 2025-01-24 by slimcdk