slimcdk/esphome-custom-components #13: Crashes with two TMC2209

jonootto, 2025-01-19:
I have manged to fairly reliably cause crashes on different boards when defining two TMC2209, as in this sketch, and tested on esp32-c3-devkitm-1 esp32dev. If you comment out the either stepper, the crashes stop. These even happen when the board is not connected to anything.

```
esphome:
  name: tmc2209-skylight
  friendly_name: tmc2209 skylight

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    
logger:
  baud_rate: 0
    
packages:
  wifi: !include common/wifi.yaml
  options: !include common/options.yaml
  web: !include common/web.yaml

wifi:
  manual_ip:
    static_ip: 192.x

web_server:
  version: 3
  
external_components:
  - source: github://slimcdk/esphome-custom-components#0035ce5
    components: [ tmc2209, stepper ]

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

uart:
  - id: uarta
    tx_pin: GPIO3
    rx_pin: GPIO0
    baud_rate: 500000 # 9600 -> 500k
  - id: uartb
    tx_pin: GPIO8
    rx_pin: GPIO7
    baud_rate: 500000 # 9600 -> 500k

button:
  - platform: template
    name: Stop
    on_press:
      - stepper.stop: driver1
      - stepper.stop: driver2

  - platform: template
    name: 1000 Steps forward
    on_press:
      - stepper.set_target:
          id: driver1
          target: !lambda return id(driver1)->current_position +1000;
      - stepper.set_target:
          id: driver2
          target: !lambda return id(driver2)->current_position +1000;

  - platform: template
    name: 1000 Steps backward
    on_press:
[... 135 lines cut ...]
[13:51:18]I (280) cpu_start: Pro cpu up.
[13:51:18]I (295) cpu_start: Pro cpu start user code
[13:51:18]I (295) cpu_start: cpu freq: 160000000 Hz
[13:51:18]I (295) cpu_start: Application information:
[13:51:18]I (298) cpu_start: Project name:     tmc2209-skylight
[13:51:18]I (304) cpu_start: App version:      2024.12.4
[13:51:18]I (309) cpu_start: Compile time:     Jan 19 2025 13:49:12
[13:51:18]I (315) cpu_start: ELF file SHA256:  b59f82778e32f365...
[13:51:18]I (321) cpu_start: ESP-IDF:          5.1.5
[13:51:18]I (326) cpu_start: Min chip rev:     v0.3
[13:51:18]I (330) cpu_start: Max chip rev:     v1.99 
[13:51:18]I (335) cpu_start: Chip rev:         v0.4
[13:51:18]I (340) heap_init: Initializing. RAM available for dynamic allocation:
[13:51:18]I (347) heap_init: At 3FC98540 len 00027AC0 (158 KiB): DRAM
[13:51:18]I (353) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
[13:51:18]I (360) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
[13:51:18]I (368) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
[13:51:18]I (375) spi_flash: detected chip: generic
[13:51:18]I (379) spi_flash: flash io: dio
[13:51:18]I (383) sleep: Configure to isolate all GPIO pins in sleep state
[13:51:19]I (390) sleep: Enable automatic switching of GPIO sleep configuration
[13:51:19]I (397) app_start: Starting scheduler on CPU0
[13:51:19]I (402) main_task: Started on CPU0
[13:51:19]I (402) main_task: Calling app_main()
```

jonootto, 2025-01-19:
In the example i used: `source: github://slimcdk/esphome-custom-components#0035ce5` but it happens on the current release also

slimcdk, 2025-01-24:
Hi @jonootto,

The issue should be fixed. The fix is however on another branch for now as it includes additional changes to running multiple drivers. Please give it a try and let me know if the issue is still there.

https://github.com/slimcdk/esphome-custom-components/blob/multiple-tmc2209/esphome/components/tmc2209/README.md

jonootto, 2025-01-25:
Works! Thank you very much!

closed 2025-01-25 by jonootto