slimcdk/esphome-custom-components #16: Motor position not reporting, motors spins indefinitely

ItsRebaseTime, 2025-02-11:
Hello,
I am working on IKEA roller blinds conversion to stepper motor with TMC2209. I configured everything as in examples, but the motor keeps spinning indefinitely and the `current_position` does not update with the actual position. Is there an issue with my wiring or GPIO selection? Position was reporting before the recent update. Tested various configurations of diag/index pins, the issue persists.
My ESP32 board: https://docs.m5stack.com/en/unit/poesp32
Connections are made with perfboard and some decent soldered wires.
Here is my configuration:

```
esphome:
  name: blinds
  friendly_name: Blinds
  on_boot:
    - tmc2209.configure:
        microsteps: 1
        interpolation: False
        enable_spreadcycle: False
        tcool_threshold: 1000 # SG can be triggered when speed exceeds this value
    - tmc2209.stallguard:
        threshold: 10 # SG is triggered if the SG value is lower than 2x threshold
    - tmc2209.currents:
        standstill_mode: normal
        irun: 31 # 16 for best microstep performance (16/32 = 1/2 maximum motor current)
        ihold: 31 # should be 0 to enable passive breaking
        tpowerdown: 0
        iholddelay: 0
    # - button.press: home_button # home right after boot

ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO5

text_sensor:
  - platform: ethernet_info
    ip_address:
      name: "IP"
    mac_address:
      name: "MAC"
    dns_address:
      name: "DNS"

external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209_hub, tmc2209, stepper ]

globals:
  - id: has_homed
    type: bool
    initial_value: "true"
    restore_value: False
  - id: blinds_bottom_pos
    type: int
    initial_value: '9999'
    restore_value: True
  - id: blinds_top_pos
    type: int
    initial_value: '0'
    restore_value: True

uart:
  id: tmc2209_uart
  rx_pin: GPIO17
  tx_pin: GPIO16
  baud_rate: 115200

tmc2209_hub:
  id: tmc_hub
  uart_id: tmc2209_uart

stepper:
  - platform: tmc2209
    id: driver
    tmc2209_hub_id: tmc_hub
    diag_pin: GPIO1
    index_pin: GPIO3
    max_speed: 500 steps/s
    acceleration: 100 steps/s^2
    deceleration: 100 steps/s^2
    address: 0x00
    # rsense: 110 mOhm
    vsense: False
    on_status:
      - logger.log:
          format: "Driver is reporting an update! (code %d)"
          args: ["code"]
      - if:
          condition:
            lambda: return code == tmc2209::DIAG_TRIGGERED;
          then:
            - logger.log: DIAG_TRIGGERED
      - if:
          condition:
            lambda: return code == tmc2209::DIAG_TRIGGER_CLEARED;
          then:
            - logger.log: DIAG_TRIGGER_CLEARED
      - if:
          condition:
            lambda: return code == tmc2209::RESET;
          then:
            - logger.log: RESET
      - if:
          condition:
            lambda: return code == tmc2209::RESET_CLEARED;
          then:
            - logger.log: RESET_CLEARED
      - if:
          condition:
            lambda: return code == tmc2209::DRIVER_ERROR;
          then:
            - logger.log: DRIVER_ERROR
      - if:
          condition:
            lambda: return code == tmc2209::DRIVER_ERROR_CLEARED;
          then:
            - logger.log: DRIVER_ERROR_CLEARED
    on_stall:
      - logger.log: "Motor stalled!"
      - if:
          condition:
            lambda: |-
              return !id(has_homed);
          then:
            - stepper.stop: driver
            - stepper.report_position:
                id: driver
                position: 0
            - globals.set:
                id: has_homed
                value: "true"
            - logger.log: "Homed"

button:
  - platform: template
    id: home_button
    name: "Home"
    on_press:
      - logger.log: "Going home!"
      - globals.set:
          id: has_homed
          value: "false"
      - stepper.set_target:
          id: driver
          target: -9999999
  - platform: template
    id: set_bottom_pos_button
    name: "Set bottom position"
    on_press:
      then:
        - globals.set: 
            id: blinds_bottom_pos
            value: !lambda 'return id(driver)->current_position;'
        - lambda: |-
            ESP_LOGD("Blinds", "Setting bottom position to: %d", id(driver)->current_position);
  - platform: template
    id: set_top_pos_button
    name: "Set top position"
    on_press:
      then:
        - globals.set: 
            id: blinds_top_pos
            value: !lambda 'return id(driver)->current_position;'
        - lambda: |-
            ESP_LOGD("Blinds", "Setting top position to: %d", id(driver)->current_position);

binary_sensor:
  - platform: template
    id: homed_status_sensor
    entity_category: DIAGNOSTIC
    name: "Homed status"
    lambda: |-
      return id(has_homed);

sensor:
  - platform: template
    entity_category: DIAGNOSTIC
    id: top_position_sensor
    name: "Top position"
    accuracy_decimals: 0
    unit_of_measurement: "steps"
    lambda: |-
      return id(blinds_top_pos);

  - platform: template
    entity_category: DIAGNOSTIC
    id: bottom_position_sensor
    name: "Bottom position"
    accuracy_decimals: 0
    unit_of_measurement: "steps"
    lambda: |-
      return id(blinds_bottom_pos);

  - platform: template
    entity_category: DIAGNOSTIC
    id: stepper_position_sensor
    name: "Position"
    unit_of_measurement: "steps"
    accuracy_decimals: 0
    update_interval: 250ms
    lambda: |-
      return id(driver)->current_position;

  - platform: tmc2209
    entity_category: DIAGNOSTIC
    internal: True
    type: stallguard_result
    name: Driver stallguard
    update_interval: 250ms
```

slimcdk, 2025-02-11:
First thing that comes to my mind is wiring for INDEX pin has disconnected or shorted to gnd or vcc. IO3 is UART0 RX, right? Make sure it isn't reserved for something else. The logger for instance. Hardware logging can be disabled https://esphome.io/components/logger.html#hardware-uarts.

Which framework do you use? And when did you update the last time it worked? ESPHome updated espidf from v4 to v5 in recent update - I don't think it would affect anything, but still keep an eye out.

Can you share the full config?

ItsRebaseTime, 2025-02-11:
Thanks for your answer and support!

Yes, I had the logger line too, I know that those pins are default for the UART0 just didn't copy it to the config I added in my issue description, updated it now (some parts I left out like OTA, HA API keys and some commented out cover component lines which are not currently used).

Framework is arduino, the config is pretty much complete in the issue description.

I will check if espidf works, haven't tested with it yet.

I am on the latest esphome dev branch (2025.2.0-dev20250211), so the position worked on 2025.2.0-dev20250209. I don't see any related changes which could have affected this, maybe I should pull out the old scope and check the index line.

slimcdk, 2025-02-11:
2025.2.0-dev works with an ESP32 S3 that I got in my development setup, though I haven't tested using UART0 lines.

Config seems good. Check the index line. You should see pulses when the motor runs/moves. Let me know when you have inspected the index line.

ItsRebaseTime, 2025-02-11:
So something really weird is happening, here is what I did:
I tested the index line, it was showing nothing (i could see the uart data, so scope settings seemed OK)
I thought maybe the esp32 is holding the index line at GND, so I desoldered the wire to it, connected the ethernet, looked at the logs and there were a lot of diag trigger prints (like the diag pin was floating).
I checked if I did desoldered the right wire and I did.
Then I thought maybe I mixed up the diag and index gpio in the config so quickly swapped them around and reflashed the esp32.
The diag prints were gone and the possition is now working fine, but no stall detection (obviously cause no diag connected)
I reconnect the diag pin, check the logs - position no longer works once again, index line is at 0.
Reflashed with the swapped index and diag gpio once again - stll the same issue.

Any ideas? 😄

ItsRebaseTime, 2025-02-11:
I would guess that the limited amount of gpios I have on this board is the reason, using bootstrapping pins is not recommended in any situation.

slimcdk, 2025-02-12:
Does it happen with previous versions of ESPHome and/or previous versions of the component? You can specific git commit in the external component link.

ItsRebaseTime, 2025-02-16:
It seems it is the same with the previous versions, I tested with another board and other gpios and everything is fine now. I don't know how it worked for me previously, maybe I did not notice the position not being reported during development of my solution. maybe USAR0 pins can't be used in such a way that both of them are interruptable. I will not investigate this further since I found a cheaper POE board with plenty more gpios for my design.

Thanks for your help.
Closing this issue.

closed 2025-02-16 by ItsRebaseTime

slimcdk, 2025-02-16:
Great to hear it got resolved. And thank you very much for pizza! 🍕 

Could we get an update on your project for converting IKEA blinds?

ItsRebaseTime, 2025-02-16:
No problem, I will link my project here after it is documented, but it will be in my [turbomansion](https://github.com/ItsRebaseTime/turbomansion) repo as all the other things I am designing for my smart home. Everyone is welcome to take a look and use anything they like.

ItsRebaseTime, 2025-02-16:
Also, I just noticed that the TMC2209 driver boards I use have index pin in the complete different position than the other boards (I use MKS V2.0 boards)
[image]