slimcdk/esphome-custom-components #17: Cannot set microsteps with a substitution.

dwmw2, 2025-02-11:
This doesn't work:
```
substitutions:
  microsteps: "2"

esphome:
  on_boot:
    - tmc2209.configure:
        microsteps: ${microsteps}
```

```
Failed config

esphome: None
  name: landing-blind-1
  platformio_options: 
    board_build.flash_mode: dio
    upload_speed: 921600
  on_boot: 
    - tmc2209.configure: 
        
        Unknown value '2', did you mean '2', '32'?.
```

This is a shame because I've redone some of the other calculations to use `${microsteps}` directly, e.g.

```
    position_action:
      - lambda: |-
          int32_t wantpos = id(sensored_home_pos) + (1.0 - pos) * ${encoder_closed_pos};
          int32_t posdelta = wantpos - id(encoder)->get_state();
          int32_t stepdelta = ((posdelta * 200 * ${microsteps}) + 2048) / 4096;
          ESP_LOGI("cover", "From %f to %f (%f to %d) is delta %d, %d steps. Step %d -> %d",
                            id(pd_blinds)->position, pos, id(encoder)->get_state(), wantpos, posdelta, stepdelta,
                            id(motor)->current_position, id(motor)->current_position + stepdelta);
          id(motor).set_target(id(motor)->current_position + stepdelta);

```

slimcdk, 2025-02-11:
I've pushed a fix. Remember to set refresh interval to 0s for pulling external components

dwmw2, 2025-02-11:
Thanks, that works. Which component have I forgotten to disable refreshes for?

slimcdk, 2025-02-11:
The `external_components` component caches components for a certain time (24h I think). Setting `refresh: 0s` will force it to fetch latest

commit 8416d329 (2025-02-11): fixed #17
commit 1254f0e1 (2025-02-11): fixed #17
closed 2025-02-11 by slimcdk