slimcdk/esphome-custom-components #20: Compile fails for TMC2208

shortyjacobs, 2025-03-19:
I'm trying to use an ESP32 to run two UART channels to two Bigtree TMC2208 V3 stepsticks, and control the stepper drivers with step/dir pins.  (need to do it this way because the stepsticks don't expose the index pin).  The yaml validates fine, but upon compile to create the flashable bin, it fails.  It fails with the same error messages if I select "update wirelessly" in ESPhome as well as "update manually".  It fails with a slightly alterred error message if I use arduino framework vs idf framework, (log also attached).

Unfortunately, I'm not skilled enough to have a clue what the errors mean, but let me know if I can help, I reallllly want this to work and appreciate your efforts on the driver project!

ESPHome validate, YAML file (txt), and compile logs attached.  

[logs_ro-water-controller_validate.txt](https://github.com/user-attachments/files/19332501/logs_ro-water-controller_validate.txt)

[logs_ro-water-controller_run (1).txt](https://github.com/user-attachments/files/19332472/logs_ro-water-controller_run.1.txt)

[ro-water-controller.yaml.txt](https://github.com/user-attachments/files/19332494/ro-water-controller.yaml.txt)

[logs_ro-water-controller_run arduino_framework.txt](https://github.com/user-attachments/files/19332574/logs_ro-water-controller_run.arduino_framework.txt)

slimcdk, 2025-03-19:
Hi,

It was a major oversight from my side when stripping down tmc2209 to a tmc2208 component. It should be resolved now, but let me know if you face any other issues.

Thank you very much for including the logs.

closed 2025-03-19 by slimcdk

shortyjacobs, 2025-03-20:
Worked perfectly, (after taking forever for me to find the "refresh" flag to force the component update from github).  Thank you!