slimcdk/esphome-custom-components #22: configuration verification error when using both tmc2209 and a4988 platforms

cgprinz, 2025-04-25:
I have an ESP32S3 setup with two happily working TMC2209 (thanks!) and an experimental MKS SERVO42D, for which I use the a4988 platform for now.
I see the following error message
```
INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/revolver-s3.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1081, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1059, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 863, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 142, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 738, in run
    self.comp.final_validate_schema(conf)
  File "/data/external_components/705079b2/esphome/components/tmc2209_hub/__init__.py", line 63, in final_validate
    steppers_in_hub = [
                      ^
  File "/data/external_components/705079b2/esphome/components/tmc2209_hub/__init__.py", line 66, in <listcomp>
    if stepper[CONF_TMC2209_HUB_ID] == config[CONF_ID]
       ~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'tmc2209_hub_id'
```
as soon as I add
```
- platform: a4988
    id: mks 
    step_pin:
      number: 19
      inverted: true
    dir_pin:
      number: 20
      inverted: false
    max_speed: 400 steps/s 
```

Here is an Excerpt from the tmc / hub config
```
uart:
  - id: tmc_comms
    tx_pin: GPIO43 
    rx_pin: GPIO44 
    baud_rate: 500000

tmc2209_hub:
  id: hub1
  uart_id: tmc_comms

stepper:
  - platform: tmc2209
    id: driver1
    tmc2209_hub_id: hub1
    address: 0x00
    ...
```
What could I be doing wrong ?

slimcdk, 2025-04-25:
Hi,

Can you share the entire yaml config?

cgprinz, 2025-04-25:
[multi-platform-error.yaml.txt](https://github.com/user-attachments/files/19917570/multi-platform-error.yaml.txt)

slimcdk, 2025-04-26:
I have compiled your configuration against all big releases this year and they all compiled without issue.

Could you try cleaning build files and compile again to see if that solves it?

Thanks for the pizza!

cgprinz, 2025-04-26:
I cleaned the build files via ESPHome GUI, clicked validate on the same dropdown menu and got the same error again.
However, I may know why you don't see the error message. In the uploaded file you need to uncomment lines 650 to 658 around the a4988. I should have pointed that out. My bad.

```
#  - platform: a4988
#    id: mks 
#    step_pin:
#      number: 19
#      inverted: true
#    dir_pin:
#      number: 20
#      inverted: false
#    max_speed: 400 steps/s 
```

Here is the uncommented version:
[multi-platform-error-uncommented.yaml.txt](https://github.com/user-attachments/files/19922512/multi-platform-error-uncommented.yaml.txt)

>> Thanks for the pizza!
You deserve it.

slimcdk, 2025-04-26:
Ah, I completely overlooked that. I have a fix that I'll push to another branch for you to test. Should be up in few min.

slimcdk, 2025-04-26:
It's online in branch `fix-loading-hub`. You need to specify that branch when linking the external components

```yaml
external_components:
  source: github://slimcdk/esphome-custom-components@fix-loading-hub
  components: [ tmc2209_hub, tmc2209, stepper ]
```

cgprinz, 2025-04-26:
Confirming no more verification error(s). Also no appearant adverse effects in initial testing.
Thank you !

dvbit, 2025-05-14:
espone does not accept custom components anymore... will this be converted? thanks!

slimcdk, 2025-05-14:
Hi @dvbit,

This is already meant to be loaded as an [external component](https://esphome.io/components/external_components) and not as a 'custom component'. (in fact, it never was despite the name of the repository)

Please note the `external_component` [in the documentation](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#setup).

commit 3cca4919 (2025-05-15): Merge pull request #23 from slimcdk/fix-loading-hub

fix #22 checking for correct stepper entry
commit fc9dc788 (2025-05-15): fix #22 checking for correct stepper entry
closed 2025-05-15 by slimcdk