slimcdk/esphome-custom-components #25: Controlling two TMC2209's with a ESP32, inconsistant operation

EvanVanVan, 2025-06-15:
Hey, this project is awesome thank you for releasing it. 

I'm having issues trying to control two TMC2209's with one ESP32. They don't seem to want to perform at full speed consistently. I'll plug in the power and test the motors with a modified version of your example config. Sometimes, they initially both run at full speed for 30 seconds. Then usually at least one drops out and barely wants to turn. Sometimes neither runs well (both turning almost imperceptibly).  

FWIW, I'm using a [NEMA 17 with a 40:1 ratio Worm Gear box](https://www.omc-stepperonline.com/nema-17-stepper-motor-l-39mm-1-68a-gear-ratio-40-1-square-worm-geabox-17hs15-1684s-wg40) (Which I think would slow the operation up drastically anyway, I'm not sure if that's factoring in to my issues). Also, I did adjust the VREF with the potentiometer to 1.9V for 1.68A (based on: https://www.circuitist.com/how-to-set-driver-current-a4988-drv8825-tmc2208-tmc2209/). Does that mean I should set IRUN to 31?

(Also, stupid question, 1 microstep is "less than" 8 microsteps meaning it should move faster/further each step, correct? lol or is 1 microstep really 1/256 microsteps and less than 8/256 microsteps? I assumed `microsteps: 1` was a fullstep, but maybe it's not.) I need to play around with the microsteps, speed, etc. to find the happy medium of speed, smoothness and quietness.

Also, you have a couple of examples of the two drivers but not in the full example config. Is it really necessary to duplicate every function for each driver `id:`? (Although, I think I just noticed the ability to group `ids:` is on your To-do list if I understand it correctly?)

Thank you!

This is my config

```
external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209_hub, tmc2209, stepper ]

ota:
  platform: esphome

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  
# Enable Home Assistant API
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
web_server:

esphome:
  name: esp32-roller-shade
  friendly_name: esp32-roller-shade
  on_boot:
    - tmc2209.configure:
        id: driver1
        microsteps: 1
        interpolation: true
    - tmc2209.configure:
        id: driver2
        microsteps: 1
        interpolation: true

    - tmc2209.stallguard:
        id: driver1
        threshold: 50
    - tmc2209.stallguard:
        id: driver2
        threshold: 50
        
    - tmc2209.currents:
        id: driver1
        standstill_mode: freewheeling
        irun: 16
        ihold: 0
        tpowerdown: 0
        iholddelay: 0
    - tmc2209.currents:
        id: driver2
        standstill_mode: freewheeling
        irun: 16
        ihold: 0
        tpowerdown: 0
        iholddelay: 0

uart:
  tx_pin: 17
  rx_pin: 16
  baud_rate: 500000

stepper:
  - platform: tmc2209
    id: driver1
    address: 0x00
    max_speed: 900 steps/s
    acceleration: 1500 steps/s^2
    deceleration: 500 steps/s^2
    config_dump_include_registers: true
    rsense: 110 mOhm
    vsense: False
    index_pin: 25
    diag_pin: 26
    on_stall:
      - logger.log: "Motor stalled!"
      - stepper.stop: driver1
  
  - platform: tmc2209
    id: driver2
    address: 0x01
    max_speed: 900 steps/s
    acceleration: 1500 steps/s^2
    deceleration: 500 steps/s^2
    config_dump_include_registers: true
    rsense: 110 mOhm
    vsense: False
    index_pin: 27
    diag_pin: 14
    on_stall:
      - logger.log: "Motor stalled!"
      - stepper.stop: driver2

button:
  - platform: template
    name: Stop
    on_press:
      - stepper.stop: driver1
      - stepper.stop: driver2

  - platform: template
    name: 5000 Steps forward
    on_press:
      - stepper.set_target:
          id: driver1
          target: !lambda return id(driver1)->current_position +5000;
      - stepper.set_target:
          id: driver2
          target: !lambda return id(driver2)->current_position -5000;

  - platform: template
    name: 5000 Steps backward
    on_press:
      - stepper.set_target:
          id: driver1
          target: !lambda return id(driver1)->current_position -5000;
      - stepper.set_target:
          id: driver2
          target: !lambda return id(driver2)->current_position +5000;

number:
  - platform: template
    name: Target position
    min_value: -100000
    max_value: 100000
    step: 100
    lambda: return id(driver1)->current_position;
    update_interval: 1s
    set_action:
      - stepper.set_target:
          id: driver1
          target: !lambda "return x;"

  - platform: template
    name: Target position
    min_value: -100000
    max_value: 100000
    step: 100
    lambda: return id(driver2)->current_position;
    update_interval: 1s
    set_action:
      - stepper.set_target:
          id: driver2
          target: !lambda "return x;"

sensor:
  - platform: tmc2209
    tmc2209_id: driver1
    type: motor_load
    name: Motor load
    update_interval: 250ms
    
  - platform: tmc2209
    tmc2209_id: driver2
    type: motor_load
    name: Motor load
    update_interval: 250ms

```

EvanVanVan, 2025-06-15:
Hmm, thank you. I thought I had fixed something by upping rsense to 120 mOhm. It seemed to be a little more reliable but it still goes in and out and only a power cycle seems to fix it. I tried switching to `type: esp-idf` but it didn't seem to help much. I'm using [BIGTREETECH TMC2209 V1.3 Stepper Motor Drivers](https://biqu.equipment/products/bigtreetech-tmc2209-stepper-motor-driver-for-3d-printer-board-vs-tmc2208?srsltid=AfmBOopEieC0E0fAkdWvoB4mS5YOzwIvFxzTbK5YW6i3wbdntR4ckjUS). Sometimes they get hottish...but I also don't really want to touch them they seem to short and burn out easily lol. I've already gone through 3 during testing.

Maybe it could be signal interference? Between the ESP32 and the drivers there is a 10'6" cat6 cable to one of the drivers and 6'0" cat6 to the other. With the cat6 cable's there is 24V split across two 23AWG conductors. (I'm hoping that's safe enough, it should be based on voltage drop calculators, etc.) It is the further one that usually has the speed issue.  During testing today there sometimes seems to be some signal interference/noise with my home theater speakers.

I've added all of your example status events to the yaml file, so I'm hoping I see something there too.

TY

slimcdk, 2025-06-15:
Signal interference could be the case. Can you try with a low baud rate? Something like 115200 or 9600, which IIRC is the lowest possible for tmc2209.

EvanVanVan, 2025-06-15:
9600 didn't seem to work, gave me a communication error, 115200..worked but similarly inconsistently. And I think another driver burned out haha...It's super hot and not responding at all any more..throwing UART errors.

I might be going back to the drawing board and using step/dir. It looks like I can do that with your library? Although it still does communicate via UART (for addressing maybe?)

I was looking to go with UART because I heard it was quiet, but so far it's been loud at reasonable speeds (which again might be due to my 40:1 worm gearbox) and UART seems a little bit overkill for a simple roller shade lol.